byte4geek/SEPLOS_MQTT #1: Extending the scope to cover a second pack

championc, 2022-10-08:
The Master is always "Pack 00", but if you daisychain a second Seplos, it will be given the identity Pack 01

It would be good if the script can pull the data for Pack 0x

Great work

byte4geek, 2022-10-10:
i did think to this but i do not have a second battery pack for now, i can't test it.

championc, 2022-12-03:
Hey again,

I can test.  The packs are 00, 01, 02, ......0x

Enjoy your coffee (or beer) :-)

byte4geek, 2022-12-04:
I will try to write a script for 2 packs but I can't test it, I have only one pack.

championc, 2022-12-04:
Just a quick update.  I can run the script with ADDR=00 and later with ADDR=01 and the results are different (as I expect).  So it is correctly able to extract the results from each pack.  Of course, the best solution is to pull all results at the one moment

championc, 2022-12-04:
Further Feedback.  If I query with ADDR=02, it returns an error, which is correct.  Entering ADDR=01 should assume that you have two packs 00 & 01

So ADDR should represent the Last Pack in a setup

championc, 2022-12-04:
Further Update

I have made a crude hack version to call query_seplos_ha.sh, output all data, then run a modified query01_seplos_ha.sh on Pack01 and output all data with 01 before each data variable (so \"01lowest_cell\":\"Cell $lowcellnumb - $lowcell mV\",\ etc...)

byte4geek, 2022-12-05:
good, i can add 2 scripts one for each pack executed by the first script

championc, 2022-12-05:
UPDATE - Seplos have confirmed that you CANNOT obtain the details from a Primary Master pack when it has an additional Secondary unit in parallel.

So it's maybe best to stick with just your original script ONLY

ztardik, 2022-12-06:
@championc 
> UPDATE - Seplos have confirmed that you CANNOT obtain the details from a Primary Master pack when it has an additional Secondary unit in parallel.

Hi, can you explain in more details what they meant with this?

@byte4geek 
> good, i can add 2 scripts one for each pack executed by the first script

Could you write a short how-to on this? I'm also with dual pack configuration.

byte4geek, 2022-12-06:
You cant do this in parallel, but you can do this in sequence.

I'm writing another script that call query_seplos_ha00.sh and after call query_seplos_ha01.sh and send the data to HA in different naming.

byte4geek, 2022-12-06:
i added the scripts you simply download and add to your SEPLOS_MQTT folder.

update the user password, mqtt server ip and so on
```
chmod 700 run_bms_query_dual.sh 
chmod 700 query_seplos_ha01.sh
```
and try to run  ./run_bms_query_dual.sh

you must create all entity manually in Home Assistant like the first pack.
all mqtt entity in home assistant are called with 01_ prefix see below:

01_lowest_cell
01_lowest_cell_v
01_lowest_cell_n
01_highest_cell
01_highest_cell_v
01_highest_cell_n
01_difference
01_cell01
01_cell02
01_cell03
01_cell04
01_cell05
01_cell06
01_cell07
01_cell08
01_cell09
01_cell10
01_cell11
01_cell12
01_cell13
01_cell14
01_cell15
01_cell16
01_cell_temp1
01_cell_temp2
01_cell_temp3
01_cell_temp4
01_env_temp
01_power_temp
01_charge_discharge
01_total_voltage
01_residual_capacit
01_soc
01_cycles
01_soh
01_port_voltage

byte4geek, 2022-12-06:
If it works change the entry in crontab with:

@reboot nohup ~/SEPLOS_MQTT/run_bms_query_dual.sh &

championc, 2022-12-06:
The problem with two packs in parallel is that, when querying Pack00 and Pack01, the results are the same.  The only solution is to make 2 x Master.

Personally, I am now going to change my cells configuration to 2p16s and only use 1 x Seplos BMS, and sell the other one

byte4geek, 2022-12-06:
Sorry but I don't understand, if you use the vendor tool, you can switch between each pack.
This assume that you have address 00 and 01 and the cable connected PC -> pack 00 -> pack 01, this is correct?

championc, 2022-12-09:
Seplos say that if you have a Pack01, Pack02, Pack03 .......  then you can NEVER read the details of Pack 00 via RS485 :(

They say that the dip switch must be changed to all off, to read the Pack00 details

byte4geek, 2023-01-12:
> Seplos say that if you have a Pack01, Pack02, Pack03 ....... then you can NEVER read the details of Pack 00 via RS485 :(
> 
> They say that the dip switch must be changed to all off, to read the Pack00 details

Do you have any update regarding 2 packs?

MrAlgo, 2023-01-17:
I also wanted to monitor 2 packs / BMS but it’s not possible via their software as confirmed by Seplos and presumably no other software due to hardware restrictions of the BMS? This video explains it further. 

https://youtu.be/liu86NoFiQc

MrAlgo, 2023-01-18:
I wonder if the data the screens receive is also RS485? Could you tap into the data sent to them for those of us with two packs?

ztardik, 2023-01-19:
The screen is probably rs232, that is the cheapest option for short links. But there is no option to edit parameters on the screen.
One other option is to try listening on the internal 485 bus. Maybe the master shares it's data with the slaves.
The last option is to use CAN bus. I've got an adapter but could not test it until April.

MrAlgo, 2023-01-19:
That would work for me as I only need to be able to monitor both packs remotely when away rather than change any settings.

Presumably, I would just need an RS232 to USB adapter for the hardware but could this bash script be modified to work with the screens output?

Got to say that I am a novice with all of this and my skills stop at implementing what others have created and some slight modification or addition of code to make it work with my HA instance so I'm not going to be much help in that department, but I can test. 

Seplos have just come out with a Bluetooth version of this BMS. Some new hardware on the new revision of screens and some on the BMS itself.

I spoke to Seplos to see if replacing my current screens would give me BT read-only access and they confirmed it would so I ordered 2 which are en route to me now. This at least solves checking both packs while I'm within BT range via their app. Just need to sort the same but when I'm away.

markvk42, 2023-02-06:
Ive just been watching the off grid garage set of vids on this, as far as im aware if you have the same type of BMS and you are not trying to use can bus then you can monitor all of the batteries within their software. If you are using Can then you cant monitor the master only the slave. To get around this they introduce the BT version, but its stupidly buggy!

byte4geek, 2023-02-06:
i removed the script for 2 batteries since seplos made this questionable choice that not to allow tracking of multiple packages connected together, i am very disappointed with this choice of seplos.

markvk42, 2023-02-07:
Ok Thanks, so how would i monitor 3 banks? 3 ras pies or i pi running containers some how?

ztardik, 2023-02-07:
@byte4geek Have you looked into the CAN protocol? Maybe it would be a viable option to just listen on the CAN bus in case of more than one pack. 
I have MKS CANable Pro that I bought some time ago for 25 Eur, I can test it in April.

bbbowden, 2023-02-09:
Concerning multiple packs...  We are connecting multiple batteries in parallel and utilizing the CANBUS to aggregate the data from those batteries into a pack message.  In the CANBUS message, the high and low cell and which battery the cell is in is indicated in the CANBUS message.  Unfortunately, the individual cell data is not relayed in the CANBUS message.

So, I am looking at not using a "Master Battery" (meaning that its address is set to 0) and rather gathering the RS485 data myself and then generating the CANBUS message to be used by the inverter.  This does mean that the maximum number of batteries I can have in parallel is 15, but that shouldn't be a problem.

I'm in the process of modifying these scripts to sequentially query each subsequent battery.  Using the Seplos BMS software, you can see how the query message changes i.e.:
Battery 1: ~20014642E00201FD35$
Battery 2: ~20024642E00202FD33$
etc...
The things that change are characters 3 and 4, characters 13 and 14, and then the checksum.

So, modifying the script shouldn't be too hard.

Now my question:  

Looking at the communication log in the Seplos BMS software, there are two commands sent for each pack query: Get pack remote communication data and Get pack telemetry data.  Do you have the protocol description of the response to the Get pack remote communication data message?  It's not critical (I don't think) but I'm curious what is in there.

By the way, thank you for parsing the pack telemetry data.  That is a huge help in my efforts.

byte4geek, 2023-02-10:
I found only this document on internet about the protocol, try to find all information inside.
https://www.dropbox.com/s/myvjpi74prjkcgz/SEPLOS%20BMS%20Communication%20Protocol_V2.0.pdf?dl=0

bbbowden, 2023-02-15:
I have created a barebones Node Red flow to query the Seplos BMS across multiple batteries.  Once you set the number of batteries to query, set the Multi Inject Time stamp to re-occur at your sample interval.  The battery data comes out on debug 7 and can be used however you want (i.e. inject into MQTT).  

You will need to have the Buffer Parser node installed in your Node Red instance.  It's the only non-standard node used in this flow.

Hope someone finds this helpful.

[SeplosBMS_Multi_Query.zip](https://github.com/byte4geek/SEPLOS_MQTT/files/10743921/SeplosBMS_Multi_Query.zip)

bbbowden, 2023-02-15:
Opps, missed a couple of nodes (the ones to set the number of batteries) when I exported the flow.   
[SeplosBMS_Multi_Query_Set_Number of batteries.zip](https://github.com/byte4geek/SEPLOS_MQTT/files/10744144/SeplosBMS_Multi_Query_Set_Number.of.batteries.zip)

markvk42, 2023-02-15:
> Opps, missed a couple of nodes (the ones to set the number of batteries) when I exported the flow. [SeplosBMS_Multi_Query_Set_Number of batteries.zip](https://github.com/byte4geek/SEPLOS_MQTT/files/10744144/SeplosBMS_Multi_Query_Set_Number.of.batteries.zip)

bbbowden and all others. Thanks for your work, when i get my batteries im certain ill be back to ask questions!

pedro37, 2023-03-07:
> Opps, missed a couple of nodes (the ones to set the number of batteries) when I exported the flow. [SeplosBMS_Multi_Query_Set_Number of batteries.zip](https://github.com/byte4geek/SEPLOS_MQTT/files/10744144/SeplosBMS_Multi_Query_Set_Number.of.batteries.zip)

> I have created a barebones Node Red flow to query the Seplos BMS across multiple batteries. Once you set the number of batteries to query, set the Multi Inject Time stamp to re-occur at your sample [...]

Thanks, is it possible for a step by step guide?

navesile, 2023-03-13:
> Opps, missed a couple of nodes (the ones to set the number of batteries) when I exported the flow. [SeplosBMS_Multi_Query_Set_Number of batteries.zip](https://github.com/byte4geek/SEPLOS_MQTT/files/10744144/SeplosBMS_Multi_Query_Set_Number.of.batteries.zip)

awesome work. I got it up but keep getting weirdness for cid. any insight? 
[image]

navesile, 2023-03-27:
thanks [bbbowden](https://github.com/bbbowden) for the  the heavy lifting. I figured it out.  messed it up, cut things, added things, sent to db, just started to play with grafana. 
[image]
here is my flow. 
[28V2battery.zip](https://github.com/byte4geek/SEPLOS_MQTT/files/11080292/28V2battery.zip)

byte4geek, 2023-10-24:
Monitor 2 or more batteries is possible only via CAN protocol

closed 2023-10-24 by byte4geek

navesile, 2023-10-24:
i I don't think so much details is available in can