byte4geek/SEPLOS_MQTT #2: Error Code 05

tim-mills, 2022-11-27:
I’m desperate for this to work, just keep getting an error code 5, anything I can try?

tim-mills, 2022-11-28:
Thanks for getting back to me. That's what I've been trying, Just comes straight back with Error Code 05. Just struggling to see how I can debug the issue, so any pointers are welcome. 

[image]
[image]

byte4geek, 2022-11-28:
check if you have address 1 (battery number 1) in the battery configuration

tim-mills, 2022-11-28:
Battery configuration, as in via the Seplos software?

tim-mills, 2022-11-28:
Just to confirm, all my DIP switches are OFF, which I think means it's set to single pack. I've tried setting DIP 1 to ON, which means it's 1st pack, although this didn't seem to work and also meant my Solis Inverter got a CAN fail.

byte4geek, 2022-11-28:
i seen you did ask for this issue on the original discussion of the original script creator:
https://diysolarforum.com/threads/simple-seplos-bms-protocol-decode-bash-script.34993/
wait for the answare in the forum i don't know error 5 says.

tim-mills, 2022-11-28:
No worries - I've just noticed you just put a wrapper around the script. I'll wait and see if they get back to me on there, I wonder if there's been a shift in the different versions of the BMS.

roverga, 2022-12-29:
Great project, but I have the same problem: Error 5!

byte4geek, 2022-12-31:
> Great project, but I have the same problem: Error 5!

Sorry but regarding this error, you must ask to the author of the original script as specified 2 post up.

roverga, 2023-01-06:
I finally find a solution for the "Error Code 05" that works for me. 

I write my solution on the original discussion:
[https://diysolarforum.com/threads/simple-seplos-bms-protocol-decode-bash-script.34993/post-677063](https://diysolarforum.com/threads/simple-seplos-bms-protocol-decode-bash-script.34993/post-677063)

closed 2023-01-11 by byte4geek