byte4geek/SEPLOS_MQTT #4: Home Assistant not receiving Data

KillianMcGrath, 2022-12-05:
Hi folks,

Bit of a novice here so would appreciate the help please

- Have installed on Rpi3
- configuration.yaml is edited as per script

Communications working
`pi@raspberrypi:~ $ ~/SEPLOS_MQTT/query_seplos_ha.sh 4201
3272
3274
3274
3273
3272
3271
3270
3270
3268
3273
3273
3272
3273
3271
3273
3271
23.7
23.8
20.3
22.1
23.7
22.9
-21.52
52.35
226.93
341.83
66.3
400.00
80
100.0
52.33
pi@raspberrypi:~ $
`

- Can confirm that HA receives other MQTT data
- nohup.out file contents below
`/home/pi/SEPLOS_MQTT/run_bms_query.sh: 23: Syntax error: Bad for loop variable
/home/pi/SEPLOS_MQTT/run_bms_query.sh: 23: Syntax error: Bad for loop variable
/home/pi/SEPLOS_MQTT/run_bms_query.sh: 23: Syntax error: Bad for loop variable`

IP, Username and Password is corect
`# insert the mqtt info below
# mqtt host name
MQTTHOST=192.168.1.102
# name inserted into topic
TOPIC=seplos
# mqtt user name
MQTTUSER=username_mqtt
# mqtt password
MQTTPASWD=password_mqtt
# time to read and update datas vs mqtt server and Home Assistant
TELEPERIOD=10
# is a prefix inserted into topic, chage it if you need
id_prefix=364715398511
# Max size of the BMS_error.log and nohup.out
MAXSIZE=2000000
# Log file name
LOGNAME=~/SEPLOS_MQTT/BMS_error.log
# nohup file name for standard and error output
NOUPFILE=~/SEPLOS_MQTT/nohup.out`

- BMS Error Log

`Sun  4 Dec 13:57:46 GMT 2022 - Possible bad read data from BMS - Error 6
Sun  4 Dec 13:57:56 GMT 2022 - Possible bad read data from BMS - Error 6
Sun  4 Dec 13:58:07 GMT 2022 - Possible bad read data from BMS - Error 6
Sun  4 Dec 13:58:24 GMT 2022 - Possible bad read data from BMS - Error 6
Sun  4 Dec 13:58:34 GMT 2022 - Possible bad read data from BMS - Error 6
Sun  4 Dec 13:58:44 GMT 2022 - Possible bad read data from BMS - Error 6
                                                                         Sun  4 Dec 18:35:04 GMT 2022 - Possible bad read data from BMS - Error 2
Sun  4 Dec 18:35:05 GMT 2022 - Possible bad read data from BMS - Error 6
Sun  4 Dec 18:37:33 GMT 2022 - Possible bad read data from BMS - Error 6
Sun  4 Dec 18:37:33 GMT 2022 - Possible bad read data from BMS - Error 6
Sun  4 Dec 18:37:48 GMT 2022 - Possible bad read data from BMS - Error 2
Sun  4 Dec 18:37:52 GMT 2022 - Possible bad read data from BMS - Error 6
Sun  4 Dec 18:38:26 GMT 2022 - Possible bad read data from BMS - Error 2
Sun  4 Dec 18:38:29 GMT 2022 - Possible bad read data from BMS - Error 6
Sun  4 Dec 18:39:54 GMT 2022 - Possible bad read data from BMS - Error 6
Sun  4 Dec 18:39:57 GMT 2022 - Possible bad read data from BMS - Error 1
Sun  4 Dec 18:40:02 GMT 2022 - Possible bad read data from BMS - Error 6
Sun  4 Dec 18:45:29 GMT 2022 - Possible bad read data from BMS - Error 2
Sun  4 Dec 18:45:32 GMT 2022 - Possible bad read data from BMS - Error 6
Sun  4 Dec 18:46:35 GMT 2022 - Possible bad read data from BMS - Error 6
Sun  4 Dec 18:46:38 GMT 2022 - Possible bad read data from BMS - Error 1
Sun  4 Dec 18:46:43 GMT 2022 - Possible bad read data from BMS - Error 6
Sun  4 Dec 18:53:50 GMT 2022 - Possible bad read data from BMS - Error 6
Sun  4 Dec 18:53:54 GMT 2022 - Possible bad read data from BMS - Error 6
Sun  4 Dec 18:54:49 GMT 2022 - Possible bad read data from BMS - Error 6
Sun  4 Dec 18:54:53 GMT 2022 - Possible bad read data from BMS - Error 6
Sun  4 Dec 18:55:49 GMT 2022 - Possible bad read data from BMS - Error 2
Sun  4 Dec 18:55:52 GMT 2022 - Possible bad read data from BMS - Error 6`

f anyone could point me in the right direction that would be great thanks

byte4geek, 2022-12-06:
did you tried to update the raspberry pi os?

```
sudo apt-get update
sudo apt-get upgrade -y
```

KillianMcGrath, 2022-12-06:
The update does not work, seems like an issue with the pi as I cannot send message to HA using CLI either

pi@raspberrypi:~ $ mosquitto_pub -h 192.168.1.102 -u username_mqtt -p password_mqtt -t homeassistant/sensor/seplos_364715398511 -m "Hello"
Error: No such file or directory
pi@raspberrypi:~ $

KillianMcGrath, 2022-12-06:
Seems to be a user issue with HA, after deleting and adding MQTT user I seem to be able to get messages through
pi@raspberrypi:~ $ mosquitto_pub -d -h 192.168.1.102 -p 1883 -u user12365 -P pass123584 -t "homeassistant/sensor/seplos_364715398511" -m "Hello"
Client (null) sending CONNECT
Client (null) received CONNACK (0)
Client (null) sending PUBLISH (d0, q0, r0, m1, 'homeassistant/sensor/seplos_364715398511', ... (5 bytes))
Client (null) sending DISCONNECT
pi@raspberrypi:~ $