byte4geek/SEPLOS_MQTT #8: Esp32/8266

JoeGhost-22, 2023-04-27:
Hi, I dont know anything about codeing, but is there any way to run your code on a esp32 or 8266. ? That would help me a lot, cause I dont want to spend a lot of money for a raspi to monitor the battery pack.  I run the other code from github, but Im not satisfied with it. 
regards
Joerg.

byte4geek, 2023-04-28:
Sorry this code is not for esp32, but you can run this on an old eeepc with debian OS you can buy an old eeepc for a little money

JoeGhost-22, 2023-04-28:
Thank you for the Info. Two more questions. Can it monitor multiple packs? Are u still working on it.? I've 2 packs at the moment and plan to add a third. Do I need a  eepc for every pack?

Thank you and have a nice weekend
Best regards
Joerg

byte4geek, 2023-04-29:
Sorry i'm not working for 2 packs or more, but for the simple way you can use 2 rs485 adapter and run 2 script in parallel by changing the directory script name and changing the /dev/ttyUSB0 name in the /dev/ttyUSB1, but after this change the id_prefix=364715398511 in id_prefix=364715398512 with another number and create all entity in HA with the prefix 00 and 01 and nn

for example:

mqtt:
  sensor:

    - name: "BMS01 Cell 01"  <--- changed
      state_topic: "homeassistant/sensor/seplos_364715398512"  <--- changed
      value_template: '{{ value_json["cell01"] }}'
      unit_of_measurement: "mV"
      unique_id: "seplos_cell01_364715398512"  <--- changed
      icon: mdi:car-battery
      device_class: voltage
      expire_after: 600

closed 2023-10-24 by byte4geek