byte4geek/SEPLOS_MQTT #9: Works with Waveshare rs485 ip converter

matchermike1981, 2023-04-30:
Hi. I use waveshare rs485 to eth ip adapters for my inverter control. Can this work with one of them for my bms? I run ha on a remote pi. If not can this run on pi local to seplos with USB rs485 adapter and send data to my ha pi?

Thanks!

byte4geek, 2023-05-02:
This script send the data in MQTT, but i think any device configured in linux like /dev/xxxxxx can send it to MQTT server.

byte4geek, 2023-05-02:
Please provide your /dev/name of your RS485 remote ip adapter.

matchermike1981, 2023-05-02:
Hi. 

My RS485 adapter is this: https://www.waveshare.com/wiki/RS485_TO_ETH_(B) 

It has various configuration options. I wonder if i can use this in some way to communicate back to my HA instance so i have the stats of my seplos BMS as you do?

Many Thanks
Mike

byte4geek, 2023-05-03:
I think your adapter will have drivers to install on the remote system that will configure a device, try to rely on that device by running the script to read the data.

closed 2023-10-24 by byte4geek