byte4geek/SEPLOS_MQTT #11: Script with Seplos 3.0

otello86, 2023-09-07:
Hi,
Do you know if this script is working with the version 3.0 of the Seplos BMS ?
I bought a battery with this version of bms and for the moment i got directly this error : "Failed to read start of input char (~), read ""
I have no line with "Sending...."
I check the rs485 connector and it works fine the seplos Monitor tool

byte4geek, 2023-09-07:
Sorry but i do not have the hardware 3.0 i can't help you, in any case you can ask to the forum where the author wrote the intial script query_seplos.sh https://diysolarforum.com/threads/simple-seplos-bms-protocol-decode-bash-script.34993/page-2#post-677063

byte4geek, 2023-09-19:
Whout a board to test it is impossible for me to do the script for the version 3

closed 2023-10-24 by byte4geek