byte4geek/SEPLOS_MQTT #12: SEPLOS_MQTT

Tiqilla, 2023-10-23:
Hello byte4geek,

unfortunately I need your support with the SEPLOS_MQTT project. I don't get along with the configuration. At the following command line I didn't know exactly what to do and in which format to send the command.  It is the following line:
ssh-copy-id root@<YOUR HA IP> ---> and choose yes

Can you help me please?

Greetings Steffen

byte4geek, 2023-10-24:
Hi are you trying to install this script directly on Home Assistant? If yes, when you run ssh-copy-id you are enabling the passwordless connection from home assiatant and the ssh server component this is required to start the script from the HA automation.

run this command with ip of Home Assistant 
ssh-copy-id root@<YOUR HA IP>

for example:
 ssh-copy-id root@192.168.1.10

maybe you need to run ssh-keygen before without words (press enter to all question), this is to create a public key if you don't already have it.

Tiqilla, 2023-10-26:
Hello,
unfortunately I speak very bad English. That's why I have to translate everything with Deepl. Unfortunately it still does not work. I have attached a screenshot for you. Do you have a solution for me?

Greetings Steffen

 

Gesendet: Dienstag, 24. Oktober 2023 um 08:15 Uhr

Tiqilla, 2023-10-29:
Hi, " If yes, when you run ssh-copy-id you are enabling the passwordless connection from home assiatant and the ssh server component".

 

How can I do this. I do not know how to do this. Can you please explain it to me in more detail?

 

Best wishes Steffen

 

 
 

Gesendet: Dienstag, 24. Oktober 2023 um 07:15 Uhr

Tiqilla, 2023-11-02:
Hi,

unfortunately I still have problems with the setup. Maybe you can support me again. In the log of Advanced SSH & Web Terminal I have the following error messages. This comes since I created the ssh-key. Do you have a solution for my problem? This is an excerpt from the log:
Advanced SSH & Web Terminal

 

Excess permission or bad ownership on file /var/log/btmp

Failed password for invalid user root from 192.168.178.75 port 36984 ssh2

Excess permission or bad ownership on file /var/log/btmp

Connection closed by invalid user root 192.168.178.75 port 36984 [preauth]

Connection from 192.168.178.75 port 44478 on 192.168.178.75 port 22 rdomain ""

User root from 192.168.178.75 not allowed because not listed in AllowUsers

Failed none for invalid user root from 192.168.178.75 port 44478 ssh2

Failed password for invalid user root from 192.168.178.75 port 44478 ssh2

Excess permission or bad ownership on file /var/log/btmp

Failed password for invalid user root from 192.168.178.75 port 44478 ssh2

Excess permission or bad ownership on file /var/log/btmp

Connection closed by invalid user root 192.168.178.75 port 44478 [preauth]

Connection from 192.168.178.75 port 55062 on 192.168.178.75 port 22 rdomain ""

User root from 192.168.178.75 not allowed because not listed in AllowUsers

Failed none for invalid user root from 192.168.178.75 port 55062 ssh2

Failed password for invalid user root from 192.168.178.75 port 55062 ssh2

Excess permission or bad ownership on file /var/log/btmp

Failed password for invalid user root from 192.168.178.75 port 55062 ssh2

Excess permission or bad ownership on file /var/log/btmp

Connection closed by invalid user root 192.168.178.75 port 55062 [preauth]

Connection from 192.168.178.75 port 53262 on 192.168.178.75 port 22 rdomain ""

User root from 192.168.178.75 not allowed because not listed in AllowUsers

Failed none for invalid user root from 192.168.178.75 port 53262 ssh2

Failed password for invalid user root from 192.168.178.75 port 53262 ssh2

Excess permission or bad ownership on file /var/log/btmp

Failed password for invalid user root from 192.168.178.75 port 53262 ssh2

Excess permission or bad ownership on file /var/log/btmp

Connection closed by invalid user root 192.168.178.75 port 53262 [preauth]

Connection from 192.168.178.75 port 33120 on 192.168.178.75 port 22 rdomain ""

User root from 192.168.178.75 not allowed because not listed in AllowUsers

Failed none for invalid user root from 192.168.178.75 port 33120 ssh2

Failed password for invalid user root from 192.168.178.75 port 33120 ssh2

Excess permission or bad ownership on file /var/log/btmp

Failed password for invalid user root from 192.168.178.75 port 33120 ssh2

Excess permission or bad ownership on file /var/log/btmp

Connection closed by invalid user root 192.168.178.75 port 33120 [preauth]

Connection from 192.168.178.75 port 56390 on 192.168.178.75 port 22 rdomain ""

User root from 192.168.178.75 not allowed because not listed in AllowUsers

Failed none for invalid user root from 192.168.178.75 port 56390 ssh2

Failed password for invalid user root from 192.168.178.75 port 56390 ssh2

Excess permission or bad ownership on file /var/log/btmp

Failed password for invalid user root from 192.168.178.75 port 56390 ssh2

Excess permission or bad ownership on file /var/log/btmp

Connection closed by invalid user root 192.168.178.75 port 56390 [preauth]

Connection from 192.168.178.75 port 53486 on 192.168.178.75 port 22 rdomain ""

 

Best wishes Steffen

 
 

Gesendet: Dienstag, 24. Oktober 2023 um 07:15 Uhr

closed 2024-06-21 by byte4geek