byte4geek/SEPLOS_MQTT #13: work over tcp

xjustbmw, 2023-10-27:
can we modify this to work any a modbus -> tcp adapter over ip ?

byte4geek, 2023-10-27:
is a good idea, if you already performend the configuration using modbus adapter or similar and ,the seplos tool, tell me what mod bus adapter you used, otherwise i do not have in program to do this.

if i understand correctly we will have this configuration?

SEPLOS board <-> modbus adapter <-> TPCIP <-> modbus adatper <-> RS485 adapter <-> PC with seplos tool
or
SEPLOS board <-> modbus adapter <-> TPCIP <-> PC with driver that configure virtual RS485 port and seplos tool

xjustbmw, 2023-10-27:
yes. this is the ideea.
I am using elfin devices. (EE11 model) about 10$ each, works perfectly , 1 for the deye inverter (works directly over modbus rtu) and one for the battery
the device already does the conversion from ip:port to modbus (frame or RTU)

I am looking at this pylontech battery nodered flow here:
Is it the same ? 

https://github.com/meteosat007/solar-pylontech/blob/main/Pylontech%20Battery%20Master.json

care to look if it is the same ?

xjustbmw, 2023-10-28:
[image]
[image]

finally made it directly in nodered using telnet tcp module

closed 2024-05-20 by byte4geek