byte4geek/SEPLOS_MQTT #14: Multiple Seplos BMS

BarkinSpider, 2023-11-07:
Do you have any information about what data is returned when you have a master BMS and multiple slaves attached? I have 4 of the Seplos/Mason 48v boxes. Do you need a RS485 connection to each box or can you just query the master?

xjustbmw, 2023-11-07:
not sure you can query at all if they already use the rs485 for parallel connection. what bms version you got ?

BarkinSpider, 2023-11-07:
Hi, they have dual RS485 ports on the Mason boxes. The use the V2 BMS.
I will hook them up with my Pi4 next week and find out what happens.

See docs attached

I will be adding a Grafana interface if it works. Look at my Git code to
see what that will look like. I have coded many BMS/inverters/charge
controllers already.

What cable did you use? What is the RJ45 pinout?

Like that you got the comms working in just SH, I had to use Python up till
now.

Thanks ... Joe

xjustbmw, 2023-11-07:
I am also with seplos v2 , waiting for a neey to balance the pack. The internal passive balancer takes forever.

From what i read on an offgrid garage video, the master battery unfortunately cannot be read anymore with v2 seplos in parallel mode.

the new V3 of the seplos bms still has a parallel mode that you can use to monitor all your batteries.

A different ideea would be to monitor all of them over a wifi->bluetooth gateway.

regards,
Adrian

xjustbmw, 2023-11-07:
regarding the pinout, I used pin 7 and 8 for A and B.

riogrande75, 2023-11-29:
Playing arround with this issue as well...
If you interconnect some seplos bms' (V16 in my case), the master polls slave id's on the RS485 bus with following commands every second:
`~2001465A0000FD9D
~2001465AC04000010BF40BAD0B8D0B85FFDE12E51E2B6D60011312E900000000000003028010EFE5
~2001465A0000FD9D
~2001465AC04000010BF40BAD0B8D0B85FFDC12E51E2B6D60011312E900000000000003028010EFE7
~2001465A0000FD9D
~2001465AC04000010BF40BAD0B8D0B85FFDC12E51E2B6D60011312E900000000000003028010EFE7
~2001465A0000FD9D
~2001465AC04000010BF40BAD0B8D0B85FFDC12E51E2B6D60011312E900000000000003028010EFE7`
There is no documentation about CID2 0x5A so far, but decoding seems not too complicated.

As the master box takes over the (RS485) busmuster functionality, you cannot poll anything on that bus unless you're ok with lots of collisions.
If you are interrested in every single battery's voltage,etc., you could poll all slaves but don't get the info from the master battery.

BarkinSpider, 2023-12-07:
Hi, finally got around to trying your code. Running the query script all I can get is

Response: "~2001465A0000FD9D"

Nothing else. I using the cable supplied by seplos for PC connection. Ive tried all port and dip switch combinations.
Any ideas? Thanks … Joe.

riogrande75, 2023-12-07:
_~2001465A0000FD9D_ ist just a query command from the master battery. You obviously captured just this one.
If you can't see a reply from a slave battery, I guess there is currently no communication between your batteries.

kasperbirk, 2024-02-22:
Hello. Great job with this code! I am curious about this issue, because I just upgraded my to batteries with seplos cases and already uses Home Assistant. Is there any news about this?

ddonq1412, 2024-05-14:
Would be interested getting this to work for V2 BMSs currently have 3 units. Have knocked off the inverter Comms for slaves to get this working for home assistant.

byte4geek, 2024-05-15:
i do not have version 2, and now installed the version 3 and i use this https://github.com/syssi/esphome-seplos-bms to monitor it.

closed 2024-05-15 by byte4geek